manuelbl/ttn-esp32 #2: Same issue Mode=1 (lukecyca), ESP32 with RFM95W

H-LK, 2018-08-06:
Hi,
many thanks for this library.
I am testing with an ESP32 Pico Dev Kit and Hope RFM95W using the Heltec pin config.

I have to report the same observation. Only mode=1 works.

Changing hal_esp32.c to .mode=0 results in:
E (1128) ttn_hal: /home/hlk/esp/hello_world/components/ttn-esp32/src/hal_esp32.c:464

manuelbl, 2018-08-06:
I'm somewhat surprised as I have a RFM95W myself and it works nicely without any changes to the code. Using a DSO I've even checked the SPI signals and found them to match the SX1276 specification.

I'll will try your pin configuration and let you know about the results.

You mentioned the line hal_esp32.c:464 in your issue. All raised ASSERTs end there. Do you have any additional line information?

manuelbl, 2018-08-08:
With your pin configuration, I was able to reproduce the issue. Even though the SX1276 data sheet clearly states the SPI mode to be CPOL = 0 and CPHA = 0 (equivalent to mode 0), mode 1 reliably works across five boards I've tested.

So I've changed the SPI mode to 1 (see 5cd2c7bac4b34a102d5a9211d3f381d6ce7b59e9).

closed 2018-08-08 by manuelbl