manuelbl/ttn-esp32 #9: Getting it work with Platform IO

csvke, 2019-03-28:
I am very new to this: is there anything I should be aware of if I want to use your libraries in PlatformIO?

manuelbl, 2019-03-28:
Currently, I don't use Platform IO for ESP32 development. So I can't tell you. But I will soon try and probably add some instructions in the Wiki if anything special is needed.

manuelbl, 2019-04-19:
In the Wiki, you'll now find a dedicated guide for PlatformIO: https://github.com/manuelbl/ttn-esp32/wiki/Using-PlatformIO.

ESP-IDF is not well supported as ESP-IDF relies on *Kconfig*/*menuconfig* for configuration and PlatformIO does not support it. So I'm thinking about adding dedicated board specification to PlatformIO at a later stage.

Nevertheless, you can use PlatformIO and once you have made the correct configuration, it's a really powerful and productive tool.

closed 2019-04-27 by manuelbl