manuelbl/ttn-esp32 #12: Add TTGO T-Beam to wiki Boards and Pins page

jcwren, 2019-04-25:
Would you mind adding the TTGO T-Beam info to the `Boards and Pins` page of the wiki?

GitHub page with image, etc: https://github.com/LilyGO/TTGO-T-Beam

It may be worth noting on the wiki that the NEO-6M that purports to be on the board is usually counterfeit, and is actually a Hangzhou ZhongKe Microelectronics AT6558 part (datasheet: AT6558 BDS/GNSS Full Constellation SOC Chip Data Sheet
www.icofchina.com/d/file/xiazai/2016.../b1be6f481cdf9d773b963ab30a2d11d8.pdf), so the ublox UBX mode isn't supported. 

Here's the pin list:

```
#define TTN_SPI_HOST      HSPI_HOST
#define TTN_SPI_DMA_CHAN  1
#define TTN_PIN_SPI_SCLK  5
#define TTN_PIN_SPI_MOSI  27
#define TTN_PIN_SPI_MISO  19
#define TTN_PIN_NSS       18
#define TTN_PIN_RXTX      TTN_NOT_CONNECTED
#define TTN_PIN_RST       TTN_NOT_CONNECTED
#define TTN_PIN_DIO0      26
#define TTN_PIN_DIO1      33
```

Uses standard 40MHz system clock.

I use the ublox UBX mode in a couple products, and replaced the NEO-6M with a NEO-M8Q. I have some example code that pushes the GPS position and ESP-32 chip temperature to The Things Network. I'm fine with sharing it, but as it parses the UBX output and not NMEA, it won't be useful to anyone who doesn't replace the NEO-6M with an authentic ublox part.

manuelbl, 2019-04-25:
It's a wiki. Unless I have misconfigured it, every GitHub user should have the right to change the page.

You are very welcome to add the TTGO T-Beam to the page.

jcwren, 2019-04-25:
I don't see a way to edit it. It's quite possible I'm blindly missing it, but only option I see is to clone it locally, and GitHub doesn't support PRs for wikis.

manuelbl, 2019-04-25:
Sorry, it was restricted indeed.

I've changed it and you should now find an *Edit* button in the top right corner.

manuelbl, 2019-10-06:
I've finally added the board to the Wikil

closed 2019-10-06 by manuelbl