manuelbl/ttn-esp32 #13: Stuck at ESP_INTR_FLAG_IRAM

yene, 2019-05-12:
I was able to setup the ESP32 toolchain and run the official hello_world project.
But when running the ttn-esp32 hello_world I get the error
`'ESP_INTR_FLAG_IRAM' was not declared in this scope`

If I add the missing header it continuous with `'gpio_install_isr_service' was not declared`

What did I miss?

manuelbl, 2019-05-12:
Can you provide additional information?

- What version of the ESP-IDF library are you using?
- Which build system/IDE (make, cmake, PlatformIO) are you using?

yene, 2019-05-12:
I followed official tutorial so I guess the latest of master branch?
I am using macOS 10.14.4, GNU Make 3.81
```
Toolchain path: /Users/.../esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/.../esp/esp-idf/requirements.txt are satisfied.

App "hello_world" version: 657614e-dirty
CXX build/main/main.o
/Users/.../Developer/ESP32/heltec/hello_world/main/main.cpp: In function 'void app_main()':
/Users/.../Developer/ESP32/heltec/hello_world/main/main.cpp:66:36: error: 'ESP_INTR_FLAG_IRAM' was not declared in this scope
     err = gpio_install_isr_service(ESP_INTR_FLAG_IRAM);
                                    ^
/Users/.../Developer/ESP32/heltec/hello_world/main/main.cpp:66:54: error: 'gpio_install_isr_service' was not declared in this scope
     err = gpio_install_isr_service(ESP_INTR_FLAG_IRAM);
                                                      ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-frame-address'
make[1]: *** [main.o] Error 1
make: *** [component-main-build] Error 2
```

yene, 2019-05-12:
I fixed it by using the latest stable release instead of master branch

```
cd ~/esp/esp-idf
git checkout tags/v3.1.4
git clean -df
git submodule update --recursive
```

closed 2019-05-12 by yene