manuelbl/ttn-esp32 #14: Problems getting hello_world running on ESP-IDF 4.0

zdila, 2019-08-19:
For ESP-IDF I had to perform following changes for hello_world example to compile:

- add `#include <freertos/semphr.h>` to `hal_esp32.h`
- add `#include "driver/gpio.h"` to `main.cpp`

Thanks for awesome library!

manuelbl, 2019-08-19:
Thanks for the info.

Most likely I will not upgrade the library to ESP-IDF 4.0 until it's officially released. But with your changes, anyone can upgrade earlier.

zdila, 2019-08-19:
Maybe a macro could be used now:
```c
#if IDF_MAJOR_VER >= 4
#include <freertos/semphr.h>
#endif
```

```c
#if IDF_MAJOR_VER >= 4
#include "driver/gpio.h"
#endif
```

manuelbl, 2019-08-20:
ESP-IDF 4.0 is not stable yet. So any such code change might not work anymore with later development releases or with the final 4.0 release. I'm very reluctant to implement it.

zdila, 2019-08-20:
OK, no problem, just giving some hints ;-)

manuelbl, 2019-12-03:
The *dev* branch now contains a major update to the library including compatibility with the upcoming version 4 of ESP-IDF.

manuelbl, 2020-08-01:
Release 3.2.0 is has been released and is based on ESP-IDF 4.0.x

closed 2020-08-01 by manuelbl