manuelbl/ttn-esp32 #15: ABP support

ninnux, 2019-10-08:
Hi,
Is it possible to have ABP support? 
Probably it is possible to write an ABP client example using lmic low level directive, like LMIC_setSession.
I tried to do it with no success, but probably you could have success.
Can you help me? Using ABP can solve the other issue of deep sleep session.
Thank you
Nino

manuelbl, 2019-10-08:
Currently, I'm in the process of making major internal changes to the library (simplified timer handling, detailed logging, compatibility with IDF 4). Once they are stable, I can look into ABP.

Unfortunately, my knowledge of ABP is limited. What's the motivation for using ABP instead of OTAA? Can you provide a description how you would like to use in practice? How do you plan to provision the keys? How do you plan to set the frame counter if the device reboots? Do you have a good sketch that shows how to configure LMIC for ABP (for all regions)?

ninnux, 2019-10-10:
I would use session parameters (netid,devaddr,nwkskey,appskey) to use ABP after OTAA join (and getSession to get all those parameters). This way it would be possible to receive downlink messages  after a deepsleep. It is necessary to save frame counters to do it working.
I'm trying with your example code calling directly getSession and setSession LMIC function and it seems to work. When I have a working code I will share it.

ninnux, 2019-10-12:
It works!!
Here an example I'm testing: https://github.com/ninnux/esp-idf_deepsleep_lorawan_bme280/blob/otaa%2Babp/main/main.cpp

I used RTC memory for session data and frame counter:
https://github.com/ninnux/esp-idf_deepsleep_lorawan_bme280/blob/1aba3137045b73041ab6176928a6e8bccf58453a/main/main.cpp#L61-L67
Here the ABP code:
https://github.com/ninnux/esp-idf_deepsleep_lorawan_bme280/blob/1aba3137045b73041ab6176928a6e8bccf58453a/main/main.cpp#L317-L327
Let me know how you thinking about.

manuelbl, 2020-07-31:
I fully understand the need for deep sleep. However, for proper operation and for being a good citizen in the LoraWAN / TTN network, many additional parameters need to be saved. Otherwise the device is stuck at a frequency and a data rate when in fact it should be hopping between frequencies, adapt the data rate and obey other commands from the gateway. I therefore consider your and many similar implementations a hack.

Unfortunately, the underlying LMIC library does not yet support saving and restoring all relevant parameters. Until it does or I find another suitable library, I'm not going to provide hooks for the hack.

manuelbl, 2021-09-29:
Despite the missing support in the underlying LMIC library, I've added support for going into deep sleep and for powering off the ESP32 while retaining the state of the current TTN session.

It should considerably simplify these use cases. It takes care of all the details like when can the device go to sleep (if MAC commands need confirmation), what needs to be saved, handling the clock (very important) etc.

Any feedback is appreciated.

closed 2023-08-01 by manuelbl