manuelbl/ttn-esp32 #18: Disabling ADR mode in uplinks

alafi, 2019-10-30:
My plan is to use this library with Europe configuration on nodes connecting to LoRaServer, not to TTN.
I faced horrible delays when sending uplinks to LoRaServer while everything was OK when connecting to TTN.
My first insight after troubleshooting this issue is that LoRaServer sends unconfirmed downlinks after each uplink. I am not an expert but after reading some posts and tracing the message contents, I found that each uplink coming out from the node has the ADR flag enabled which cause LoRaServer to reply with ADR message, while on the contrary TTN seems not adhering to the same rule.

Now I am not sure about this, but I tried to resolve the issue by inserting a call to `LMIC_setAdrMode(false)` in `TheThingsNetwork.cpp` before the join process starts and in deed the issue seems to be resolved.

I wonder if this is the right solution for it or maybe this functionality should be exposed in a public method. Appreciate your feedback.

Just another note, my node is TTGO T-BEAM connected through a Dragino LG308 gateway.

manuelbl, 2019-10-30:
I will have a look at it once LMIC version 3.x is fully integrated. However, non-TTN stuff isn't top of my priorities.

alafi, 2019-10-31:
It would be nice also to have all LMiC API's exposed in public methods as is.
Thank you anyways for the great efforts you're putting into this port and and looking forward for your future follow up.

manuelbl, 2019-10-31:
The *ttn-esp32* library is not designed as a thin LMIC wrapper but as an easy to use TTN library. The interface should stay the same even if major changes are made to LMIC or LMIC is replaced with a different implementation. For that reason I've taken great care not to expose any LMIC parts in the library interface.

Furthermore, access to LMIC's data structures and use of its functions is only possible in certain LMIC states (usually in callbacks or when LMIC is idle). Due to the use of a separate task for the LMIC core, direct access to LMIC features make no sense. Read values would not be consistent and operations altering LMICs state would take effect with delay.

Therefore, I have no plans whatsoever to make LMIC's API available. I will add more features, but with a high-level interface, independent of LMIC.

manuelbl, 2020-08-01:
Enabling/disabling ADR has now been implemented in release 3.2.0

closed 2020-08-01 by manuelbl