manuelbl/ttn-esp32 #20: change datarate

haorfani, 2020-03-02:
Hi all,

I am trying to change the datarate, when I was using the lmic library from arduino IDE I was using the     LMIC_setDrTxpow(DR_SF10,14); command to change txpower and data rate. However when I am doing the smate with ttn-esp32 do not seem to change according to the metadata in ttn console.

Best,
Haris

manuelbl, 2020-03-02:
Using LMIC functions directly will not reliably work. I will probably add a dedicated function for this.

What's your use case for setting the data rate explicitly (and not relying on the automated setting)?

haorfani, 2020-03-02:
I want to evaluate different datarates a synchronization  scheme I developed.

haorfani, 2020-05-20:
Hi again, in the same manner is there any way to change the rxDelay? Because using LMIC function (i.e. LMIC.rxDelay = 2;) does not make any difference as far as I can see.

manuelbl, 2020-05-20:
If you are looking for a library to work outside the LoRaWAN / TTN standards, then *ttn-esp32* is probably not the library for your project.

If you modify *rxDelay*, the device won't be able to receive downlink messages in TTN environment anymore. So it will never be exposed via an API.

haorfani, 2020-05-21:
OK, thanks for your response. 

Is there any chance that you know any library that would fit more to my project then?

Best,
Haris

manuelbl, 2020-05-21:
I don't have a good overview of LoRa libraries. A Google search reveals [esp32-lora-library](https://github.com/Inteform/esp32-lora-library). It's described as "Low-level Lora (SX127_) library for the ESP-32 IDF". So that might be a fit.

Or [lmic-esp-idf](https://github.com/TobleMiner/lmic-esp-idf)...

I haven't used any of these libraries. So I can't make a recommendation.

haorfani, 2020-05-21:
thanks a lot!

closed 2020-05-21 by haorfani