manuelbl/ttn-esp32 #21: send receive issue

cslehel, 2020-03-29:
Hi, thank you for this great library. 
I have a problem. 
I am using two TTGO v1 boards without the display. They can communicate with each other i tried with the arduino-LoRa library. 
I configured the parameters and uploaded the send_recv example but they are stuck at the Joining loop.

Joining...
I (402) ttn: event EV_JOINING
I (8292) ttn: event EV_TXSTART
I (13342) ttn: event EV_RXSTART
I (14342) ttn: event EV_RXSTART
I (14582) ttn: event EV_JOIN_TXCOMPLETE

I tried to send and receive with one board (Arduino) and the second with the send_recv example (IDF) but no message received. 

I am using IDF 4.0. 
Do i need a real connection to the TTN network ?
Can i send messages between boards without connecting to the TTN network ?

Thank you for taking me into consideration.

manuelbl, 2020-03-29:
This library implements the TheThingsNetwork / LoraWAN protocol, i.e. a protocol where devices talk to gateways and gateways to application servers. So it requires a TTN registration and application server software. It's not suitable for device to device communication.

cslehel, 2020-03-30:
Thank you very much for your response. 
Your library is the only one i found which works with IDF 4. 
Do you know any library which works with IDF 4 and can be used for device to device communication ?

manuelbl, 2020-03-30:
I don't know about any suitable library for IDF. But my overview is limited. So it might exist.

closed 2020-03-30 by cslehel