manuelbl/ttn-esp32 #24: Problems with gpio_install_isr_service / ESP_INTR_FLAG_IRAM

marcel151, 2020-05-20:
I got the same problem as in the following closed issue:
https://github.com/manuelbl/ttn-esp32/issues/13

"'ESP_INTR_FLAG_IRAM' was not declared in this scope [64,11]"
"'gpio_install_isr_service' was not declared  in this scope [64,36]"

I am using PlatformIO so I don't know how to use another version of ESP-IDF.

My system:
macOS 10.15.4
VSCode 1.45.1
PlatformIO Core 4.3.3 - Home 3.2.2
ESP-IDF 3.40000.200303

I am relatively new to PlatformIO, so I hope I didn't do anything wrong. I just followed your Get-Started Guide (https://github.com/manuelbl/ttn-esp32/wiki/Get-Started) and did anything you mentioned.

manuelbl, 2020-05-20:
I'm afraid I can't help you with PlatformIO. I'm struggling myself to get it working. As far as I can tell, the PlatformIO builder is not capable of using third-party components.

If you want to experiment yourself, you definitely want to use the *dev* branch of this library.

vrees, 2020-05-29:
I had same and some more problems to get ttn-esp32 running using platformio and esp-idf. But now its working. See working example see here: https://github.com/vrees/Esp32LoraBoard

To get it running I had to change following:
- copied dev branch of ttn-esp32 in lib subdir of my platformio project
- in main.cpp #include "driver/gpio.h" in oder to get rid of error mentioned by @marcel151 
- Due to a problem wth AES encryption **undefined reference to `lmic_aes_encrypt** I had to switch to software encryption by changing 2 files. This works for me just now. Does somebody know how to use hardware encryption correctly?
`
~/work/git-repos/Esp32LoraBoard/lib/ttn-esp32$ git diff src/lmic/config.h
diff --git a/src/lmic/config.h b/src/lmic/config.h
index 8673b46..0e3bb37 100644
--- a/src/lmic/config.h
+++ b/src/lmic/config.h
@@ -135,7 +135,7 @@
-// #define USE_ORIGINAL_AES
+#define USE_ORIGINAL_AES
`

`
~/work/git-repos/Esp32LoraBoard/lib/ttn-esp32$ git diff src/aes/lmic_aes.c
diff --git a/src/aes/lmic_aes.c b/src/aes/lmic_aes.c
index 2a5bca3..c576cf7 100644
--- a/src/aes/lmic_aes.c
+++ b/src/aes/lmic_aes.c
@@ -260,7 +260,7 @@
 u4_t os_aes (u1_t mode, xref2u1_t buf, u2_t len) {
         while( (signed char)len &gt; 0 ) {
             u4_t a0, a1, a2, a3;
-            u4_t t0, t1, t2, t3;
+            u4_t t0, t1=0, t2, t3;
             u4_t *ki, *ke;
`
- added shutdown method to reduce deep sleep current of Lora chip
`
void TheThingsNetwork::shutdown()
{
    ttn_hal.enterCriticalSection();
    LMIC_shutdown();
    waitingReason = eWaitingNone;
    if (lmicEventQueue != nullptr)
    {
        xQueueReset(lmicEventQueue);
    }
    ttn_hal.leaveCriticalSection();
}
`

My system:
VSCode 1.45.1
Espressif 32 Platform displayed in 'PIO Home' 1.12.1
Linux Mint

manuelbl, 2020-08-02:
I've updated the wiki Page [Using PlatformIO](https://github.com/manuelbl/ttn-esp32/wiki/Using-PlatformIO) with the latest information about using PlatformIO.

closed 2020-08-02 by manuelbl