manuelbl/ttn-esp32 #28: Library fixed to full LoRaWAN spectrum

Doom4535, 2020-06-15:
From what I can tell, the library does not allow the end user to configure the supported LoRaWAN channels.  This is a problem for us in the U.S. at least, where we have 64 standard LoRaWAN channels, but The Things Network only operates on 8 channels (and most budget consumer gateways only support up to 8 channels anyway) as it leads to an automatic uplink packet loss of 75% (random channel uplink from a pool of 64 channels against a max supported 8 channels on the gateway).

On a side note, this library also works with Chirpstack and other LoRaWAN infrastructure as it is built against LMIC, which might be nice to reference in the README/project notes that it isn't exclusive to TTN.

P.S.  Thanks @manuelbl for this library, I appreciate the time you've spent to port LMIC to the [esp-idf](https://github.com/espressif/esp-idf)

dgashby, 2021-07-25:
Just wondering if the SubBand spamming described in the tail end of Issue #27 could be fixed by the updated channel hopping mechanism described in the v3.99.0-2 (Pre-release of V4) release of the underlying MCCI LoRaWAN LMIC?

manuelbl, 2021-07-28:
An upcoming release will automatically select sub-band 2 for regions that use sub-bands (incl. U.S.). If that's not sufficient, a sub-band can explicitly selected with a new method.

Furthermore, LMiC v4.0.1 (prerelease) has been integrated so the improved channel hopping is included as well.

The changes are in the master branch. I would appreciate if you could give it a try.

closed 2023-08-01 by manuelbl