manuelbl/ttn-esp32 #33: TTN ISR executes hal_ticks() from the flash

vladcebo, 2021-01-02:
Need to add IRAM_ATTR to hal_ticks() to prevent it executing from flash region. Otherwise the application will crash in case of OTA updates or other things that disables the flash.

cyberman54, 2021-01-03:
@vladcebo Regarding to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/general-notes.html#dma-capable-requirement, don't we also need DRAM_ATTR for buffers of SPI communication in [hal-esp32](https://github.com/manuelbl/ttn-esp32/blob/master/src/hal/hal_esp32.cpp#L169) ?

vladcebo, 2021-01-03:
@cyberman54, not sure about this one. The spi transmit is eventually called in lmic/radio.c file and the buffer passed is in the LMIC struct (LMIC.frame) which is declared as non-const static, so it's already in DRAM:
```
#define DEFINE_LMIC  struct lmic_t LMIC
```
I think the DRAM_ATTR is required when we want to send something like a const char* from the flash. For DMA it also requires it to be word aligned, and I think that's already the case (at least I never saw any errors on that so far).

manuelbl, 2021-01-03:
Thanks a lot for the PR and the other input. I will check the code and analyze if the relevant data structures end up in DRAM as expected.

manuelbl, 2021-01-05:
I've move a buffer used for SPI transmission from the stack to static allocation to be on the safe side even if PSRAM is used. But generally all relevant data structures should be in DRAM and have fixed allocation.

PR #34 (2021-01-05): Prevent hal_ticks() from going into .text region
Fixes #33 

This will fix an issue when the flash is disabled (due to OTA update for example) and the cpu crashes with cache disabled access from ISR (when TTN executes)
closed 2021-01-05 by manuelbl