manuelbl/ttn-esp32 #36: Join problems on TTN v3?

mbatini, 2021-04-22:
Hi,
just revamping my Heltec-based sensors that were up and running under TTN v2. Moving to TTN v3 I have a registered gw but devices are unable to ttn.join.

Any hints/direction on documentation or similar problems?

I register device with joinEUI as all 0x00 as stated, devEUI and AppKey are ok.

I use this library under PlatformIO/VSCode. 

Thank you in advance
Mauro

mbatini, 2021-04-23:
> I will soon have a v3 gateway. Can you try with joinEUI 0x00000001 until
> then?

Ok, it works !!!!

Also, this morning on TTN there is a naming change I think. Registering new
devices I've seen AppEUI again and not more joinEUI. A little confusion.

Thanks for help but I would like, if possible, to know details of the
solution. Maybe you will update the issue or do you need me to provide
code, tests, etc?

BTW: I'm using a rak7248G gw. It's old but it is possibl to setup a server
and just giving eu1.cloud.thethings.network was enough.

Bye
Mauro

-- 
Cordiali saluti / Best Regards / Mit freundlichen Gruessen / A Bientot /
Saludos / Met vriendelijke groet /
Med venlig hilsen / �dv�zlettel / Me Filikous Xairetismous / Saygilar /
Zaijian / Srdacne Pozdrave /
Cumprimentos / H�lsningar / Terveisin

Mauro Batini
Subject Matter Expert
Sistemi Informativi S.p.A. - An IBM Company
Mobile:335.7506786
Tel. 06.502921 - Mobile:335.7506786

manuelbl, 2021-04-23:
I haven't investigated it yet. But a similar questions was raised for *arduino-lmic* (the underlying library used by this library): https://github.com/mcci-catena/arduino-lmic/issues/690#issuecomment-792336037

nemccarthy, 2021-07-20:
did you have any luck with this one, now that v2 is no longer accepting new devices

manuelbl, 2021-07-28:
The master branch contains a fix that now support AppEUI/JoinEUI consisting of all 0.

I would appreciate if you give it a try.

mbatini, 2021-07-30:
Hi,
thanks for info, gonna try and give feedback. I only need a little time
because involved with work. Thanks again.

Mauro

-- 
Cordiali saluti / Best Regards / Mit freundlichen Gruessen / A Bientot /
Saludos / Met vriendelijke groet /
Med venlig hilsen / �dv�zlettel / Me Filikous Xairetismous / Saygilar /
Zaijian / Srdacne Pozdrave /
Cumprimentos / H�lsningar / Terveisin

Mauro Batini
Subject Matter Expert
Sistemi Informativi S.p.A. - An IBM Company
Mobile:335.7506786
Tel. 06.502921 - Mobile:335.7506786

closed 2023-08-01 by manuelbl