manuelbl/ttn-esp32 #39: ESP32-S2 compatibility

Dylan-Pestsense, 2021-06-26:
Hi Manuel,

Is this library compatible with the ESP32-S2? 

I see no reason for it not to be, I have it compiling, however I am getting stuck at line 222 of TheThingsNetwork.cpp:
`xQueueReceive(lmicEventQueue, &event, portMAX_DELAY);`

I have managed to have 3 packets send through to the TTN but not sure what was done differently. The code wouldn't work with these lines from radio.c:

803: ASSERT((readReg(RegOpMode) & OPMODE_LORA) != 0);
925: ASSERT((readReg(RegOpMode) & OPMODE_LORA) != 0);
1162: ASSERT( i != 0 );
1372: ASSERT(0);

After commenting those lines out the code runs.

The output I am getting is this:
`--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM4 instead...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x172c
load:0x4004c000,len:0xa04
load:0x40050000,len:0x2b34
SHA-256 comparison failed:
Calculated: bb00b9c49a5696d12d64acd52be127a64e4068d644bb285e0ffebddc022cefd2
Expected: dcf2571cd019a6b73d9f7272f52a52787ec3e7936b0b7cb0e18511e1e06c4c07
Attempting to boot anyway...
entry 0x4004c1f4
I (42) boot: ESP-IDF v4.4-dev-1594-g1d7068e4b 2nd stage bootloader
I (42) boot: compile time 21:44:14
I (42) boot: chip revision: 0
I (46) boot.esp32s2: SPI Speed      : 40MHz
I (51) boot.esp32s2: SPI Mode       : DIO
I (55) boot.esp32s2: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 00100000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0a4e0h ( 42208) map
I (123) esp_image: segment 1: paddr=0001a508 vaddr=3ffbe4a0 size=01e6ch (  7788) load
I (126) esp_image: segment 2: paddr=0001c37c vaddr=40022000 size=03c9ch ( 15516) load
I (136) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=1d370h (119664) map
I (171) esp_image: segment 4: paddr=0003d398 vaddr=40025c9c size=087fch ( 34812) load
I (183) esp_image: segment 5: paddr=00045b9c vaddr=50000000 size=00010h (    16) load
I (189) boot: Loaded app from partition at offset 0x10000
I (189) boot: Disabling RNG early entropy source...
I (203) cache����~f^,W��
                        �����    : size 8KB, 4Ways, cache line size 32Byte
I (203) cpu_start: Pro cpu up.
I (215) cpu_start: Pro cpu start user code
I (215) cpu_start: cpu freq: 160000000
I (215) cpu_start: Application information:
I (219) cpu_start: Project name:     hello_world
I (224) cpu_start: App version:      3e38aeb
I (229) cpu_start: Compile time:     Jun 26 2021 15:01:18
I (235) cpu_start: ELF file SHA256:  7ad958c9a84b0a4e...
I (241) cpu_start: ESP-IDF:          v4.4-dev-1594-g1d7068e4b
I (248) heap_init: Initializing. RAM available for dynamic allocation:
I (255) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (261) heap_init: At 3FFC0F28 len 0003B0D8 (236 KiB): DRAM
I (268) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (274) spi_flash: detected chip: generic
I (279) spi_flash: flash io: dio
I (283) cpu_start: Starting scheduler on PRO CPU.
I (296) ttn_hal: IO initialized
I (296) ttn_hal: SPI initialized
I (306) ttn_hal: Timer initialized
I (326) ttn_prov: Dev and app EUI and app key saved in NVS storage
Joining...
I (336) ttn: event EV_JOINING
I (336) ttn: event EV_TXSTART
I (5676) ttn: event EV_RXSTART
I (6666) ttn: event EV_RXSTART`

Any thoughts on what could be happening here?

Cheers

Dylan-Pestsense, 2021-06-26:
Just tried again and got this:
I (296) ttn_hal: IO initialized
I (296) ttn_hal: SPI initialized
I (306) ttn_hal: Timer initialized
I (326) ttn_prov: Dev and app EUI and app key saved in NVS storage
Joining...
I (336) ttn: event EV_JOINING
I (336) ttn: event EV_TXSTART
I (5676) ttn: event EV_RXSTART
E (5676) ttn_hal: LMIC failed and stopped: ../components/ttn-esp32/src/lmic/radio.c:1067

So I commented out /lmic/radio.c:1067:
ASSERT( (readReg(RegOpMode) & OPMODE_MASK) == OPMODE_SLEEP );

I now get exactly the same result as issue Joining infinite loop #37:

I (283) cpu_start: Starting scheduler on PRO CPU.
I (296) ttn_hal: IO initialized
I (296) ttn_hal: SPI initialized
I (296) ttn_hal: Timer initialized
I (336) ttn_prov: Dev and app EUI and app key saved in NVS storage
Joining...
I (336) ttn: event EV_JOINING
I (336) ttn: event EV_TXSTART
I (5676) ttn: event EV_RXSTART
I (6676) ttn: event EV_RXSTART
I (6786) ttn: event EV_JOIN_TXCOMPLETE
I (7546) ttn: event EV_TXSTART
I (12546) ttn: event EV_RXSTART
I (13536) ttn: event EV_RXSTART
I (13656) ttn: event EV_JOIN_TXCOMPLETE
I (14826) ttn: event EV_TXSTART

DylanGWork, 2021-06-26:
I let it run with no changes, it look awhile but got this:
I (280) ttn_hal: SPI initialized
I (280) ttn_hal: Timer initialized
I (310) ttn_prov: Dev and app EUI and app key saved in NVS storage
Joining...
I (310) ttn: event EV_JOINING
I (320) ttn: event EV_TXSTART
I (5660) ttn: event EV_RXSTART
I (6650) ttn: event EV_RXSTART
I (6770) ttn: event EV_JOIN_TXCOMPLETE
I (6770) ttn: event EV_TXSTART
I (11770) ttn: event EV_RXSTART
I (12770) ttn: event EV_RXSTART
I (12880) ttn: event EV_JOIN_TXCOMPLETE
I (14290) ttn: event EV_TXSTART
I (19630) ttn: event EV_RXSTART
I (20630) ttn: event EV_RXSTART
I (20740) ttn: event EV_JOIN_TXCOMPLETE
I (21530) ttn: event EV_TXSTART
I (26530) ttn: event EV_RXSTART
I (27530) ttn: event EV_RXSTART
I (27640) ttn: event EV_JOIN_TXCOMPLETE
I (31140) ttn: event EV_TXSTART
I (36480) ttn: event EV_RXSTART
I (37480) ttn: event EV_RXSTART
I (37590) ttn: event EV_JOIN_TXCOMPLETE
I (37880) ttn: event EV_TXSTART
I (42880) ttn: event EV_RXSTART
I (43870) ttn: event EV_RXSTART
I (43990) ttn: event EV_JOIN_TXCOMPLETE
I (44120) ttn: event EV_TXSTART
I (49460) ttn: event EV_RXSTART
I (50460) ttn: event EV_RXSTART
I (50580) ttn: event EV_JOIN_TXCOMPLETE
I (51060) ttn: event EV_TXSTART
I (56060) ttn: event EV_RXSTART
I (57060) ttn: event EV_RXSTART
I (57170) ttn: event EV_JOIN_TXCOMPLETE
I (60440) ttn: event EV_TXSTART
I (65780) ttn: event EV_RXSTART
I (66780) ttn: event EV_RXSTART
I (66900) ttn: event EV_JOIN_TXCOMPLETE
I (67080) ttn: event EV_TXSTART
I (72080) ttn: event EV_RXSTART
I (73070) ttn: event EV_RXSTART
I (73190) ttn: event EV_JOIN_TXCOMPLETE
I (74990) ttn: event EV_TXSTART
I (80330) ttn: event EV_RXSTART
I (81330) ttn: event EV_RXSTART
I (81440) ttn: event EV_JOIN_TXCOMPLETE
I (82110) ttn: event EV_TXSTART
I (87110) ttn: event EV_RXSTART
I (88110) ttn: event EV_RXSTART
I (88220) ttn: event EV_JOIN_TXCOMPLETE
I (90240) ttn: event EV_TXSTART
I (95580) ttn: event EV_RXSTART
I (96570) ttn: event EV_RXSTART
I (96690) ttn: event EV_JOIN_TXCOMPLETE
I (96700) ttn: event EV_TXSTART
I (101690) ttn: event EV_RXSTART
I (102690) ttn: event EV_RXSTART
I (102810) ttn: event EV_JOIN_TXCOMPLETE
I (105840) ttn: event EV_TXSTART
I (111180) ttn: event EV_RXSTART
I (112170) ttn: event EV_RXSTART
I (112290) ttn: event EV_JOIN_TXCOMPLETE
I (112740) ttn: event EV_TXSTART
I (117740) ttn: event EV_RXSTART
I (118740) ttn: event EV_RXSTART
I (118860) ttn: event EV_JOIN_TXCOMPLETE
I (118860) ttn: event EV_JOIN_FAILED
I (122670) ttn: event EV_TXSTART
I (128010) ttn: event EV_RXSTART
I (129000) ttn: event EV_RXSTART
I (129120) ttn: event EV_JOIN_TXCOMPLETE
I (129610) ttn: event EV_TXSTART
I (134610) ttn: event EV_RXSTART
I (135600) ttn: event EV_RXSTART
I (135720) ttn: event EV_JOIN_TXCOMPLETE
I (136520) ttn: event EV_TXSTART
I (141860) ttn: event EV_RXSTART
I (142860) ttn: event EV_RXSTART
I (142980) ttn: event EV_JOIN_TXCOMPLETE
I (143000) ttn: event EV_TXSTART
I (148000) ttn: event EV_RXSTART
I (149000) ttn: event EV_RXSTART
I (149110) ttn: event EV_JOIN_TXCOMPLETE
I (151580) ttn: event EV_TXSTART
I (156920) ttn: event EV_RXSTART
I (157920) ttn: event EV_RXSTART
I (158030) ttn: event EV_JOIN_TXCOMPLETE
I (158520) ttn: event EV_TXSTART
I (163520) ttn: event EV_RXSTART
I (164520) ttn: event EV_RXSTART
I (164640) ttn: event EV_JOIN_TXCOMPLETE
I (166500) ttn: event EV_TXSTART
I (171840) ttn: event EV_RXSTART
I (172830) ttn: event EV_RXSTART
I (172950) ttn: event EV_JOIN_TXCOMPLETE
I (173780) ttn: event EV_TXSTART
I (178770) ttn: event EV_RXSTART
I (179770) ttn: event EV_RXSTART
I (179890) ttn: event EV_JOIN_TXCOMPLETE
I (182490) ttn: event EV_TXSTART
I (187830) ttn: event EV_RXSTART
I (187980) ttn: event EV_JOINED
Joined.
Sending message...
I (187980) ttn: event EV_TXSTART
I (193360) ttn: event EV_RXSTART
I (194360) ttn: event EV_RXSTART
I (194470) ttn: event EV_TXCOMPLETE
Message sent.

DylanGWork, 2021-06-27:
Problem solved. I added:

bool TheThingsNetwork::joinCore()
{
    if (!provisioning.haveKeys())
    {
        ESP_LOGW(TAG, "Device EUI, App EUI and/or App key have not been provided");
        return false;
    }

    ttn_hal.enterCriticalSection();
    xQueueReset(lmicEventQueue);
    waitingReason = eWaitingForJoin;
    LMIC_startJoining();
    ttn_hal.wakeUp();
    ttn_hal.leaveCriticalSection();
    selectSubBand(1); //Added here
    TTNLmicEvent event;
    xQueueReceive(lmicEventQueue, &event, portMAX_DELAY);
    return event.event == eEvtJoinCompleted;
}

After using the extra code added in #35

Cheers

closed 2021-06-27 by DylanGWork