manuelbl/ttn-esp32 #41: undefined reference to `ttn_get_rf_settings'

vrees, 2021-08-19:
I got following error when trying to compile latest master branch version:

```
Executing task: platformio run 
Processing ttgo-lora32-v1 (platform: espressif32; board: ttgo-lora32-v1; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v1.html
PLATFORM: Espressif 32 (3.3.1) > TTGO LoRa32-OLED V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Warning! Starting with ESP-IDF v4.0, new project structure is required: 
https://docs.platformio.org/en/latest/frameworks/espidf.html#project-structure
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ttn-esp32> 4.0.0-pre+sha.8fa345a
Building in release mode
Linking .pio/build/ttgo-lora32-v1/firmware.elf
/home/vrees/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/ttgo-lora32-v1/lib470/libttn-esp32.a(TheThingsNetwork.o):(.literal._ZN16TheThingsNetwork13getRFSettingsE13TTNRxTxWindow+0x0): undefined reference to `ttn_get_rf_settings'
/home/vrees/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/ttgo-lora32-v1/lib470/libttn-esp32.a(TheThingsNetwork.o): in function `TheThingsNetwork::getRFSettings(TTNRxTxWindow)':
/home/vrees/work/git-repos/wasserstand-harry-b/.pio/libdeps/ttgo-lora32-v1/ttn-esp32/src/TheThingsNetwork.cpp:17: undefined reference to `ttn_get_rf_settings'
collect2: error: ld returned 1 exit status
*** [.pio/build/ttgo-lora32-v1/firmware.elf] Error 1
============================================================================== [FAILED] Took 7.82 seconds ==============================================================================
Der Terminalprozess "platformio 'run'" wurde mit folgendem Exitcode beendet: 1.
```

I checked, that **TheThingsNetwork.h** also includes **ttn.h** ant that line 670 has the method defined:
``` ttn_rf_settings_t ttn_get_rf_settings(ttn_rx_tx_window_t window);```

So normally it should compile. What is wrong?

**platformio.ini:**

```
[env:ttgo-lora32-v1]
platform = espressif32
board = ttgo-lora32-v1
framework = espidf

lib_deps =
  https://github.com/manuelbl/ttn-esp32.git
  
monitor_speed = 115200
monitor_port = /dev/ttyUSB0
upload_port = /dev/ttyUSB0

debug_tool = esp-prog
debug_init_break = tbreak setup

build_flags =
    -Wno-expansion-to-defined
    -D DISABLE_PING
    -D DISABLE_BEACONS
    -D LMIC_DEBUG_LEVEL=0
    -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS
    -D CFG_sx1276_radio=1
    -D CFG_eu868=1
    -D CONFIG_TTN_LORA_FREQ_EU_868=y
    -D CONFIG_TTN_RADIO_SX1276_77_78_79=y
    -D CONFIG_TTN_PROVISION_UART_DEFAULT=y
    -D CONFIG_TTN_PROVISION_UART_NUM=0
    -D CONFIG_TTN_TIMER_1_GROUP_0=y
    -D CONFIG_TTN_SPI_FREQ=10000000
    -D CONFIG_TTN_BG_TASK_PRIO=10  
```

manuelbl, 2021-08-19:
Thanks for reporting this.

`ttn.h` contains the function declaration. But the function definition has
a typo. See https://github.com/manuelbl/ttn-esp32/blob/master/src/ttn.c#L346

I'll fix it within 48h.

manuelbl, 2021-08-20:
It's now fixed in the master branch

closed 2023-08-01 by manuelbl