manuelbl/ttn-esp32 #43: ttn_hal: LMIC failed and stopped

BryanMM, 2021-10-04:
Greetings, i've been working on an program that mounts an WIFI AP for a period of time and then it reads some sensors to finally send the information to TTN. 
I've been getting the following error once the device tries to join ttn:
`ttn_hal: LMIC failed and stopped: ../programs/weatherStation/radio.c:1166`
Is there any special consideration that i gotta have while trying to use the esp32's WiFi and the sx1276?
Thanks in advance.

manuelbl, 2021-10-05:
Are you using OTAA? Do you call one of the *join* functions/methods?

BryanMM, 2021-10-05:
Yes, i'm currently using as base the deep sleep example you added few days ago.

BryanMM, 2021-10-05:
this it what it does once the join is called:
```
Joining...  
I (34063) ttn_hal: IO initialized
I (34063) ttn_hal: SPI initialized
I (34073) ttn_hal: Timer initialized
E (34083) ttn_hal: LMIC failed and stopped: ../programs/weatherStation/radio.c:1166
```

manuelbl, 2021-10-05:
I'm assuming this error happens when TTN is used for the first time and not when the device resumes from deep sleep.

The error could indicate that the communication between the ESP32 and the SX1276 chip isn't working properly. Are you using the correct pin configuration? Have you checked it with a simple program like the *hello_world* example?

BryanMM, 2021-10-05:
Checking the code i realized i was touching the RST pin within the wifi's AP, commented that and it fixed it.