manuelbl/ttn-esp32 #48: Downlink messages with payload type JSON arrive empty.

Fagonar96, 2022-03-21:
Greetings, i've been working on a program that is responsible for sending data frames that I get from the BLE scan to TTN. The program works great, so I implemented that the program can receive downlinks to change the operation mode of the program. When the payload type of these downlinks come in bytes I can get it just fine. However, if I change the payload type to JSON, it seems to obtain the message on the port that I assigned but the message is empty (length = 0 bytes).  Do you know what may be causing this? Or is it not implemented yet?
[image]

Thanks in advance.

manuelbl, 2022-03-22:
How do you schedule the downlink message? Did you schedule it in the TTN console?

If so, you are probably expecting that you can select *JSON*, enter a JSON message and have it sent unchanged to the LoRa device. I don't fully understand the TTN console. But my understanding is that a payload formatter needs to be created. It will encode the JSON message into binary form. Then the encoded payload will be transmitted. I'm guessing that you don't have a payload formatter. Thus the payload cannot be created and an empty message is sent.

Fagonar96, 2022-03-22:
Thanks for your reply. Correct, I schedule the downlinks through the TTN console. I noticed that TTN has Javascript payload formatters that allow you to write your own functions to encode or decode messages. So I'm going to try it this way. Thanks.

closed 2022-03-22 by Fagonar96