manuelbl/ttn-esp32 #49: ttn_hal error

isuarezbel, 2022-03-23:
I've been stuck at this point for a long time. Any idea why this might be happening?

> I (206) cpu_start: Project name:     hello_world
> I (211) cpu_start: App version:      1
> I (216) cpu_start: Compile time:     Mar 23 2022 10:09:12
> I (222) cpu_start: ELF file SHA256:  aaced9e8575a2a3f...
> I (228) cpu_start: ESP-IDF:          v4.4-dirty
> I (233) heap_init: Initializing. RAM available for dynamic allocation:
> I (240) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
> I (246) heap_init: At 3FFC1C60 len 0003A3A0 (232 KiB): DRAM
> I (253) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
> I (259) spi_flash: detected chip: generic
> I (264) spi_flash: flash io: dio
> I (268) cpu_start: Starting scheduler on PRO CPU.
> I (280) ttn_prov: DevEUI, AppEUI/JoinEUI and AppKey saved in NVS storage
> Joining...
> I (280) gpio: GPIO[20]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (290) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (300) gpio: GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (310) gpio: GPIO[37]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (320) ttn_hal: IO initialized
> I (320) ttn_hal: SPI initialized
> I (330) ttn_hal: Timer initialized
> E (340) ttn_hal: LMIC failed and stopped: ../components/ttn-esp32/src/lmic/radio.c:1206
> 

Thank you very much for your great work. An esp-idf library for lorawan is really appreciated.

manuelbl, 2022-03-23:
Which version of the ttn-esp32 code are you using? It doesn't seem to be the latest master.

isuarezbel, 2022-03-24:
Thank you for the response. I am using the latest code (I downloaded it again just in case) with esp-idf 4.4. However, I think the problem may be that I am using an esp32-s2 which has some specific SPI particularities. I am not able to read the regversion at 0x42 with the ttn-esp32 readReg, but I can this way:

```
>  // Initialize SPI bus
>     spi_bus_config_t spi_bus_config = {
>         .miso_io_num = 10,
>         .mosi_io_num = 9,
>         .sclk_io_num = 3,
>         .quadwp_io_num = -1,
>         .quadhd_io_num = -1,
>     }; 
>     esp_err_t err = spi_bus_initialize(SPI2_HOST, &spi_bus_config, SPI2_HOST);
>     ESP_ERROR_CHECK(err);
> 
>     spi_device_interface_config_t devcfg = {
>         .clock_speed_hz = 1 * 1000 * 1000, // Clock out at 10 MHz
>         .mode = 3,                         // SPI mode 3
>          //        .address_bits = 8,
>         .command_bits=8,
>         .spics_io_num=20,               //CS pin
>         .queue_size=1,                          
>         //        .cs_ena_posttrans = 2                   //Keep the CS low 
>     };
>     ESP_ERROR_CHECK( spi_bus_add_device(SPI2_HOST, &devcfg, &adc_spi) );
> 
>     spi_transaction_t t;
>     memset(&t, 0, sizeof(t));
>     t.cmd = (0x42 & 0x7f);
>     t.length = 8;
>     t.flags = SPI_TRANS_USE_RXDATA;
>     //t.flags = SPI_TRANS_USE_RXDATA | SPI_TRANS_USE_TXDATA;
>     ESP_ERROR_CHECK( spi_device_polling_transmit(adc_spi, &t) );
>     
>     ESP_LOGI(TAG, "LoRa chip version = 0x%02x\n", t.rx_data[0]);
> 
> 
>     ESP_ERROR_CHECK(spi_bus_remove_device(adc_spi));
>     ESP_ERROR_CHECK(spi_bus_free(SPI2_HOST));
> 
> 
```

manuelbl, 2022-03-24:
I'm still confused regarding what code you are using since the error message "LMIC failed and stopped" is caused by an ASSERT statement. And at ../components/ttn-esp32/src/lmic/radio.c:1206, there is no ASSERT statement in the latest code (see https://github.com/manuelbl/ttn-esp32/blob/master/src/lmic/radio.c#L1206).

But the most likely cause is indeed that the SPI communication with the LoRa chip is not working. I haven't tested in an ESP32-S2 yet. I wasn't aware it is peculiar in that regard. Do you have any additional information?

isuarezbel, 2022-03-24:
I have probably made some debug comments and the lines have been modified. It corresponds to the ASSERT on line 1166 of the original code.

In the esp32-sw datasheet:

ESP32-S2 family features four SPI interfaces (SPI0, SPI1, SPI2 and SPI3). SPI0 and SPI1 can only be configured
to operate in SPI memory mode; SPI2 can be configured to operate in SPI memory and general-purpose SPI
modes; SPI3 can only be configured to operate in general-purpose SPI mode

Also from the spi master example here:

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_master/hd_eeprom#connections

In the code they use SPI2_HOST to use FSPI on the ESP32-S2

manuelbl, 2022-03-26:
The master branch now contains a fix for the SPI communication. I've successfully verified it with an ESP32-S2.

The main problem was that the chip select (aka NSS) went low at the same time as the SPI clock started for the next transaction (instead of some 10 ns earlier). That's now fixed.

Additionally, I've changed some defaults in the examples so they are more likely to work both for the ESP32 and ESP32-S2.

I would appreciate it if you could try the latest code from the master branch.

isuarezbel, 2022-03-28:
It works great now on the ESP32-S2 with a RFM95. Thank you again for the library and for the great work!

closed 2022-03-28 by isuarezbel