manuelbl/ttn-esp32 #50: ADR TXPower doesn't change

DylanGWork, 2022-05-05:
I've started testing ADR and I can see the MAC commands from the network server changing the TXPower, however the TXPower appears to stay at the maximum setting.

Am I able to print out MAC downlinks to see what's coming in?

Cheers,
Dylan

manuelbl, 2022-05-05:
Hi Dylan

I propose you experiment with the following macros:

```
LMIC_ENABLE_event_logging=1

LMIC_DEBUG_LEVEL=1

LMIC_DEBUG_LEVEL=2
```
and then watch for `scan_mac_cmds_link_adr` and `applyAdrRequests`.

It is a functionality that this library fully relies on the LMIC library.

DylanGWork, 2022-05-05:
This output is from Debug level 2:

ADR 1:
`
I (8467) ttn: event EV_TXCOMPLETE
14881690: engineUpdate, opmode=0x910
Message of 0 bytes received on port 0:MAC Attempt 4  <<------------ MAC Recieved
Message sent.
14912148: engineUpdate, opmode=0x110
I (8957) ttn: event EV_TXSTART
14912299: TXMODE, freq=917400000, len=16, SF=9, BW=125, CR=4/5, IH=0
15233119: setupRx1 txrxFlags 0x21 --> 01
I (14087) ttn: event EV_RXSTART
start single rx: now-rxtime: 11
15233762: RXMODE_SINGLE, freq=925100000, SF=9, BW=500, CR=4/5, IH=0
rxtimeout: entry: 15236764 rxtime: 15233738 entry-rxtime: 3026 now-entry: 10 rxtime-txend: 311126
15295368: setupRx2 txrxFlags 0x1 --> 02
I (15087) ttn: event EV_RXSTART
start single rx: now-rxtime: 11
15296011: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
rxtimeout: entry: 15302662 rxtime: 15295988 entry-rxtime: 6674 now-entry: 10 rxtime-txend: 373376
15302700: processRx2DnData txrxFlags 0x2 --> 00
15302791: processDnData_norx txrxFlags 00 --> 20
I (15207) ttn: event EV_TXCOMPLETE
15303312: engineUpdate, opmode=0x900
I (15287) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Var reset check state1 3
`

ADR 2:
`
31097671: engineUpdate, opmode=0x908
I (2937) ttn: event EV_TXSTART
31097840: TXMODE, freq=918200000, len=33, SF=9, BW=125, CR=4/5, IH=0
31423779: setupRx1 txrxFlags 0x20 --> 01
I (8147) ttn: event EV_RXSTART
start single rx: now-rxtime: 10
31424422: RXMODE_SINGLE, freq=927500000, SF=9, BW=500, CR=4/5, IH=0
31429116: decodeFrame txrxFlags 0x1 --> 21
31429140: Received downlink, window=RX1, port=-1, ack=0, txrxFlags=0x21
I (8237) ttn: event EV_TXCOMPLETE
31429324: engineUpdate, opmode=0x910
Message of 0 bytes received on port 0:MAC Attempt 4 <<----- ADR/MAC Received
Message sent.
31459782: engineUpdate, opmode=0x110
I (8727) ttn: event EV_TXSTART
31459933: TXMODE, freq=917200000, len=16, SF=8, BW=125, CR=4/5, IH=0
31776240: setupRx1 txrxFlags 0x21 --> 01
I (13777) ttn: event EV_RXSTART
start single rx: now-rxtime: 10
31776883: RXMODE_SINGLE, freq=924500000, SF=8, BW=500, CR=4/5, IH=0
rxtimeout: entry: 31779694 rxtime: 31776860 entry-rxtime: 2834 now-entry: 10 rxtime-txend: 311126
31838490: setupRx2 txrxFlags 0x1 --> 02
I (14777) ttn: event EV_RXSTART
start single rx: now-rxtime: 10
31839133: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
rxtimeout: entry: 31845784 rxtime: 31839110 entry-rxtime: 6674 now-entry: 9 rxtime-txend: 373376
31845821: processRx2DnData txrxFlags 0x2 --> 00
31845907: processDnData_norx txrxFlags 00 --> 20
I (14897) ttn: event EV_TXCOMPLETE
31846428: engineUpdate, opmode=0x900
I (14977) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Var reset check state1 3
`

I'll have a look through the LMIC library as well to see if anybody has experienced this from the main github.

DylanGWork, 2022-05-05:
I don't appear to be able to build with LMIC_ENABLE_event_logging=1:

Getting a lot of these types of errors:
c:/users/dylan/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/ttn-esp32/libttn-esp32.a(ttn.c.obj):(.literal.event_callback+0x14): undefined reference to `ttn_log_event'
c:/users/dylan/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/ttn-esp32/libttn-esp32.a(ttn.c.obj):(.literal.ttn_configure_pins+0x0): undefined reference to `ttn_log_init'
c:/users/dylan/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/ttn-esp32/libttn-esp32.a(ttn.c.obj): in function `event_callback'

cyberman54, 2022-05-05:
look here https://github.com/mcci-catena/arduino-lmic/issues/21

manuelbl, 2022-05-05:
@DylanGWork I'm surprised you are getting these errors. It successfully builds when I add the following lines to the top most `CMakeLists.txt`:

```
add_definitions(
    -DLMIC_ENABLE_event_logging=1
    -DLMIC_DEBUG_LEVEL=2
)
```

Additionally, I don't fully understand the ADR TX power management. The way I read the code, the transmitted TX power is now used as the upper limit. But I might be wrong...

closed 2023-08-01 by manuelbl