manuelbl/ttn-esp32 #51: Linker error due to missing extern "C"  in void app_main() {} in file main.cpp

mike6000, 2022-05-16:
Got a linker error related to main.cpp
I changed main.cpp to have extern "C" directives 
`extern "C" void app_main() {} `
The linker error disappeared.

manuelbl, 2022-05-16:
Which web page and/or code example are you referring to? As far as I can see, all C++ samples use the `extern "C"` declaration.

mike6000, 2022-05-16:
I just followed the instructions here: https://github.com/manuelbl/ttn-esp32/wiki/Using-PlatformIO and compiled as instructed.
I realize now that the app_main() probably may have been generated by platformio and therefore is missing the extern "C" syntax.
I didnt come so far as to test the examples yet. Just checked that the main skeleton code compiled. 
Perhaps the doc in the wiki should be updated with that the users should try one of the examples?

mike6000, 2022-05-23:
Closing. Hopefully this info will help others.

closed 2022-05-23 by mike6000