manuelbl/ttn-esp32 #52: Not possible to navigate menuconfig menu when using platformio

mike6000, 2022-05-16:
Navigating menuconfig was not possible inside platformio. Non of the keys worked.
[image]
By running pio run -t menuconfig in a cmd window outside platformio the navigation keys worked.

manuelbl, 2022-05-16:
This is a known incompatibility between Visual Studio Code and ESP-IDF. See e.g. https://github.com/espressif/vscode-esp-idf-extension/issues/6

Several workarounds are mentioned. But I'm not sure if they are all compatible with PlatformIO. Espressif recommends to use the ESP-IDF extension instead. But that's a non-PlatformIO as well.

mike6000, 2022-05-17:
My workaround was to open a separate cmd window in Windows. Then run `pio run -t menuconfig` outside platformio. The menuconfig navigation seems to work ok by this methos. 
I close this issue now, it is meant to help others that might experience the same problem when following the platformio tutorial on ttn-esp32 wiki.

closed 2022-05-17 by mike6000