manuelbl/ttn-esp32 #53: esp-idf 5.0 issue

alkosa, 2022-05-23:
Hi.

This library does not compile under esp-idf 5.0. I have cloned the latest source code. I try to compile the monitoring example. And I have got this error message:

```bash
ttn-esp32/src/hal/hal_esp32.c:22:10: fatal error: esp_timer.h: No such file or directory
 #include "esp_timer.h"
          ^~~~~~~~~~~~~
```
Does this library work under esp-idf 5.0 or I need to use previous esp-idf? Or How can I compile this library properly?

Thanks for your help

manuelbl, 2022-05-23:
Version 5.0 is still under development. No version has been officially released yet, not even an early release candidate. Thus, I haven't cared testing the library with it yet.

Please downgrade to an officially released version such as ESP-IDF v4.4.1.

manuelbl, 2022-09-21:
It has now been fixed in the master branch.

closed 2022-09-21 by manuelbl