manuelbl/ttn-esp32 #56: No TTN option in component config anymore

Kientopf, 2022-09-02:
There is no option "The Things Network" in esp.idf menuconfig as it should be.
See the Get Started text: "In the menu, select Component config ---> and then select The Things Network ---> (at the very bottom)."
[image]

manuelbl, 2022-09-03:
In order to reproduce the problem, can you provide more information regarding your environment: ESP-IDF version, operating system etc.

Kientopf, 2022-09-03:
Thank you very much for the quick reply. I just managed to solve the problem by myself, it was a stupid path problem (there was a space in the path).

closed 2022-09-03 by Kientopf