manuelbl/ttn-esp32 #61: How to add a region  CN470 band code？

zhangyyhh, 2023-02-20:
Hello, can you add region CN470 band code to the source code?

manuelbl, 2023-02-20:
Hi

This library builds upon the MCCI Catena [arduino-lmic](https://github.com/mcci-catena/arduino-lmic) library. In order to support the CN470 region, it needs to be added to that library first.

You could get in contact with the maintainers of that library, or even contribute the necessary changes yourself (also see https://github.com/mcci-catena/arduino-lmic/blob/master/HOWTO-ADD-REGION.md).

zhangyyhh, 2023-02-21:
Thank you very much for your answer.

If Lora nodes join the network through ABP instead of OTAA, is it supported in ttn-esp32 code?

manuelbl, 2023-02-21:
The library is specifically designed for The Things Network (TTN). The lastest TTN version only supports OTAA. So that's what this library supports. ABP is not supported.