manuelbl/ttn-esp32 #62: Linkcheck

DylanGWork, 2023-02-23:
Hi,

We have been running lorawan compliance tests on the FW and I've followed the advise given in this slide:
https://lora-alliance.org/wp-content/uploads/2020/11/lora_alliance_certification_deep_dive.pdf

It was still vague in what was required but iterative testing has gotten us through most of it.

The key issue will be the Linkcheck requirements.

Looking at the main LMIC library it appears there isn't the API/function for a linkcheck call.
However there are some people [talking/using](https://github.com/mcci-catena/arduino-lmic/issues/838) the in-built compliance testing, I'm not sure how to activate that to test it?

So, two questions:
1. Is there a way to manually request a linkcheck? Or even a guiding start on how I might.
2. How do I start/use the in-built compliance functionality.

manuelbl, 2023-02-23:
I haven't dealt with compliance tests yet. And I can't guarantee that it works with LMIC. But the basic code seems to be in place.

From the comments at https://github.com/manuelbl/ttn-esp32/blob/281ba52155ebb51975db9e60add196b2bf4df4a5/src/lmic/lmic_compliance.c#L90 it would seem that only minor modifications are needed to activate it.

Likely, a minor modification is needed at https://github.com/manuelbl/ttn-esp32/blob/master/src/ttn.c#L537. It would probably look like this:

```
void message_received_callback(void *user_data, uint8_t port, const uint8_t *message, size_t message_size)
{
    if (LMIC_complianceRxMessage(port, message, message_size) == LMIC_COMPLIANCE_RX_ACTION_PROCESS) {
        ttn_lmic_event_t result = {
            .event = TTN_EVENT_MESSAGE_RECEIVED, .port = port, .message = message, .message_size = message_size};
        xQueueSend(lmic_event_queue, &result, pdMS_TO_TICKS(100));
    }
}
```

Why don't you give it a try?

DylanGWork, 2023-03-02:
Thanks for the quick reply.

Looks like it works but is having an issue with the power cycle potentially.

After the downlink activates test mode I put the device to sleep (just how it runs currently, if I need to change that for test mode I will) using the function ttn_prepare_for_deep_sleep(); .

When it wakes back up I get:
Transmission failed.

Let me know if there is anything else you think I should try, I'll be trying a few things over the next few days.

DylanGWork, 2023-03-02:
I just wrapped a while loop around the transmission process and it's started working, a few notes:
 - Could be just how I'm doing things, but the payload does not automatically change to the correct payload (downlink counter)
 - Doesn't change to the correct port number, need to put that in yourself
 - Need to add LMIC_reset() to evJoinCommand in lmic_compliance.c

I've finally got around to forking your repo properly, if I get things fixed I can attempt to do a pull request for the changes if you'd like and join the ranks of contributor.

DylanGWork, 2023-03-14:
I appear to have things working yet we are having trouble with a Redwoodcomms compliance tester, unfortunately i don't have access to the compliance tester so I'm not able to diagnose easily.

I have:

- Added the linkcheck code in and tested it through the LNS and it works as expected
- Added a downlink counter to reset when a downlink is received so that it exits testing mode after 192 frames (this is required by standard but I don't see it in the code and it continued past 192 frames when I let it run)
- Needed to add LMIC_reset() to the evJoinCommand in lmic_compliance.c however this creates an issue where it is both in the join process and in testing mode

Problem still occurring is:

- evJoinCommand needs to stay in testing mode once joined (it seems to enter app state after joining)
- Transmits too often, seems to be every 1-2 seconds whereas it needs to be every 5 seconds

I can likely figure out the evJoinCommand issue myself this week but it would be great to know where is the most appropriate place to allow for a 5 second delay before re-transmission so that the other rx windows are available for the compliance testing (as far as I believe).
Cheers,
Dylan

manuelbl, 2023-03-14:
Hi Dylan

Several things you mention are probably better discussed with the Terry Moore on [mcci-catena/arduino-lmic](https://github.com/mcci-catena/arduino-lmic). He's far more knowledgeable about both the relevant code and compliance testing.

I don't really understand why you added a transmission loop, are calling LMIC_reset(), are looking for a 5 second delay etc. But it sounds as if they are workarounds for an deeper problem, which is not fully understood yet.

You have mentioned `ttn_prepare_for_deep_sleep()`. So you are using the deep sleep or power off feature of this library. If so, one of the missing pieces might be that the compliance state is not saved and restored.

Does compliance testing work without deep sleep / power off? If not, I would propose to first work on getting it to pass without deep sleep, and then focus on the deep sleep after that. I can then support you with the deep sleep feature. But when i comes to the inner details of the LMIC library and in particular the compliance testing requirements, my knowledge is quite limited.

DylanGWork, 2023-03-14:
Ok sounds good, I'll head on over there to see what they say.

I prevent using the ttn_prepare_for_deep_sleep() when entering compliance until we exit compliance mode, so it occurs without deep sleep.

DylanGWork, 2023-07-10:
Hi, 
I was able to pass most of the compliance testing when used this library without going into deepsleep.

I have a work around for now which seems to be working well enough to get me past most of the compliance testing. Will post results here if you'd like.