manuelbl/ttn-esp32 #63: AU915 exceeds Time-on-air for ADR and packet confirmation

DylanGWork, 2023-04-13:
Hi,
I have been testing my device under different fail conditions such, in this case I was testing what happens with an ACK is not received on a confirmed packet.

What occurred was that the packet of 9 bytes was sent 7 times starting at SF 10 and finishing at SF12.

The SF increase of 11 and 12 which is not allowable on AU915 (or US915 I believe).

On a similar note the 11 byte limit (or sub 400ms time-on-air) on SF10 is not taken into consideration either.

Environment
I am using the latest version of this library:
https://github.com/manuelbl/ttn-esp32
I am using ESP-IDF framework in Visual Studio Code
The region is AU915 for this issue (but I work with 868 and US915 as well)
We are using Loriot as an LNS and SX1276

What's the best to to prevent that SF exceeding it's SF10 limit?

This would need to be applied for all reasons that the SF is changed, such as ADR, linkcheck, packet confirmation.

Cheers,
Dylan

Note comments from here:
https://github.com/mcci-catena/arduino-lmic/issues/929

manuelbl, 2024-05-06:
I'm not familiar with the details of the protocol. But the proposed fix is quite convincing.

You can create a PR and I will merge it.

djorr5, 2024-05-07:
@DylanGWork as you raised it did you want a say in which solution should be considered? There are 3 that I can see. Your changes and the two that Slavendam suggested.