manuelbl/ttn-esp32 #65: SX1262 compatibility

nippoo, 2023-05-26:
Hi,

I'm trying to port this library to the SX1262 transceiver (which Semtech are touting as a replacement for the SX1278 etc).

Just wondering if anyone has tried to do this before?

cyberman54, 2023-05-26:
This project is based on LMIC library. LMIC has no radio driver for SX1262. Thus, to get this project working with SX1262 you need to write a new or adapted radio.c for LMIC which covers the capabilities of SX1262 chip.

manuelbl, 2025-05-26:
I will add add it sooner or later. But currently I'm mainly work on other stuff.

Unfortunately, I cannot give specific advice unless I invest time analyzing the LMIC changes. This takes time as well and will have to wait too.