manuelbl/ttn-esp32 #69: ttn_hal LMIC failed and stopped error on ESP32S3 with RFM95W

Gaurang-1402, 2023-07-06:
Hi, I am using a [XIAO Sense ESP32S3](https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/) board with a RFM95W lora module. On running the hello_world example, I get the following error

```
I (346) ttn_prov: DevEUI, AppEUI/JoinEUI and AppKey saved in NVS storage
Joining...
I (346) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (356) gpio: GPIO[1]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (366) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:1 
I (376) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:1 
I (386) ttn_hal: IO initialized
I (386) ttn_hal: SPI initialized
I (386) ttn_hal: Timer initialized
E (396) ttn_hal: LMIC failed and stopped: /home/esp/Desktop/examples/components/ttn-esp32/src/lmic/radio.c:1166
```

I am running the following pin configuration

```
// Pins and other resources
#define TTN_SPI_HOST      SPI2_HOST
#define TTN_SPI_DMA_CHAN  SPI_DMA_DISABLED
#define TTN_PIN_SPI_SCLK  7
#define TTN_PIN_SPI_MOSI  9
#define TTN_PIN_SPI_MISO  8
#define TTN_PIN_NSS       0
#define TTN_PIN_RXTX      TTN_NOT_CONNECTED
#define TTN_PIN_RST       1
#define TTN_PIN_DIO0      4
#define TTN_PIN_DIO1      3
```
I am pretty confident the pin configuration is not an issue because I used the same pinconfig for P2P lora communication with another library and it worked successfully.

I have also set my menuconfig according to your guidance [here](https://github.com/manuelbl/ttn-esp32/wiki/Boards-and-Pins#hope-rfm95w-868-mhz-module-sx1276:~:text=the%20SX1276%20chip.-,PIN%20configuration,the%20module%20needs%20to%20be%20wired%20up%2C%20different%20pin%20configurations%20are,-possible.%20This%20one)

I scanned similar issues #67 and  #49 and used the latest version of ttn-esp32 from branch master but it doesn't fix my problem. I tried hello_world_in_c also as an example but it gave me the same error.

My esp-idf is v4.4

I'd greatly appreciate your help!

manuelbl, 2023-07-10:
This error indicates that the SPI communication between the ESP32 and the RFM95W is not working correctly.

In the XIAO Sense ESP32S3 documentation, I don't see where the GPIO0 would be available. So the following line is suspicious:

```
#define TTN_PIN_NSS       0
```

Gaurang-1402, 2023-07-11:
Gosh such a silly mistake. I was looking at digital pin numbers instead of GPIOs. You were correct about my mistake. Thank you so much for this library and pointing out this mistake - highly appreciate it!

closed 2023-07-11 by Gaurang-1402