manuelbl/ttn-esp32 #70: ttn_set_subband function no longer functions

DylanGWork, 2023-07-10:
Hi,

I have updated to the most recent build and noticed that my join process was taking longer, it would appear that it is no longer selecting the correct subband. I've had a look through the changes and haven't been able to locate the issue.

I can switch between the old version and it works perfectly each time and the new version does not.

I tried switching back to the old version of:
lmic_channelshuffle.c
lmic_us_like.c
lorabase.h

They seemed to have some possible changes related to this but no luck.

For example, I am selecting subband 2 (ttn_set_subband(2);) and getting the following logs:

```
Joining...
318100: engineUpdate, opmode=0x4
318234: TXMODE, freq=915800000, len=23, SF=10, BW=125, CR=4/5, IH=0
start single rx: now-rxtime: 10
652563: RXMODE_SINGLE, freq=925100000, SF=10, BW=500, CR=4/5, IH=0
rxtimeout: entry: 656011 rxtime: 652537 entry-rxtime: 3474 now-entry: 9 rxtime-txend: 311126
start single rx: now-rxtime: 10
714811: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
rxtimeout: entry: 721460 rxtime: 714787 entry-rxtime: 6673 now-entry: 10 rxtime-txend: 373376
721504: engineUpdate, opmode=0x4
727347: engineUpdate, opmode=0x4
727475: TXMODE, freq=915900000, len=23, SF=8, BW=500, CR=4/5, IH=0
start single rx: now-rxtime: 10
1040403: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
rxtimeout: entry: 1043116 rxtime: 1040378 entry-rxtime: 2738 now-entry: 9 rxtime-txend: 311126
start single rx: now-rxtime: 10
1102652: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
rxtimeout: entry: 1109302 rxtime: 1102628 entry-rxtime: 6674 now-entry: 9 rxtime-txend: 373376
1109353: engineUpdate, opmode=0x4
1172716: engineUpdate, opmode=0x4
1172846: TXMODE, freq=917200000, len=23, SF=10, BW=125, CR=4/5, IH=0
```

Where the last line is correctly within subband 2 and joins succesfully.

manuelbl, 2023-07-10:
I have a good idea which change could be related to this problem. Can you add the code you use for the initialization (mainly the `ttn_xxx` calls up to `ttn_join()`?

DylanGWork, 2023-07-11:
I tried adding it to the ttn_join, then tried adding them along different lines of the join_core, issue still remains.

Let me know your thoughts and I can check it out.

manuelbl, 2023-08-01:
Unfortunately, I'm unable to reproduce it. Are you using it in combination with deep sleep or power off?

When you refer to the *old version*: What exact version is it?

It would be very helpful if you provided a short test program to reproduce it.