manuelbl/ttn-esp32 #76: Newest version of ESP-IDF fails during build process

djorr5, 2024-05-05:
Hi, 

First of all thanks for this, was exactly what I was looking for. Took a long time scouring the internet to find this but when I did it was a great relief.

just as an FYI the latest build of ESP-IDF fails to compile. I used v5.2.1 to try the hello_world project and encountered the following error:

[image]

not sure where or why you would be calling this part of their library but thought I would let you know. I reverted back to 5.0.6 and it works no problem.

manuelbl, 2024-05-05:
Thank you for reporting it.

The problem seems to be known: https://github.com/espressif/esp-idf/issues/12180. But there is no obvious action.

I wasn't able to reproduce it. What operating system are you using? What chip (ESP32, ESP32S3, ESP32C3 etc.) are you targeting?

_esp_lcd_ is a component that is pulled in by default. But it isn't actually needed. So if you exclude it, it might work. To do so, add the following line to the top-level `CMakeLists.txt`:

```
list(APPEND EXCLUDE_COMPONENTS "esp_lcd")
```

djorr5, 2024-05-05:
Ok understood. I will play around in the latest esp-idf if I have time and test it further. 

Using windows 11 esp-idf in 5.2.1 in powershell, targeting esp32s3 custom board.

djorr5, 2024-05-06:
I tried again and it completed successfully without any changes done on v5.2.1. Code runs on my board all good.

closed 2024-05-06 by djorr5